neu-rah/PCINT #2: ATmega2561 Conflict - PCINT

janioanselmo, 2017-11-24:
Good afternoon, I made a small change in the library - "pcint.h" to receive the mapping of the ATmega2561. However, I notice conflict with the library "SoftwareSerial.h", in some research, I realized that it is "common" interruption library conflicts with the library "SoftwareSerial.h". Are there any alternatives? Here's some important information:

**ADD Mapping - ATmega2561:**
```
#elif (defined (__ AVR_ATmega2561__))
    const uint8_t pcintPinMap [1] [8] PROGMEM = {{8,9,10,11,12,13,14,15}};
```
**Package used:**
https://github.com/MCUdude/MegaCore

neu-rah, 2017-11-24:
yes, its a known problem. PCINT management cant be shared. Because this libraries have to take control of a shared handler and demultiplex the interrupt.
However this library uses only 3 ports you can use software serial on a port not used by this library i think.
Otherwise only using the same PCINT controller.
A remote possibility would be to cascade the handler. This ould be better having the software serial as the master controller because communications are timing sensitive.

janioanselmo, 2017-11-24:
Hi, @neu-rah 
Excellent speed in response! Excellent explanation! Of course, I've only used your library and it works perfectly! Thank you. Sorry to bother you with that question. I hope my mapping to the ATmega2561 is correct. Good weekend!

neu-rah, 2017-11-24:
you are welcome :D
and if you wish to do a pull-request with that map I would be pleased to accept it

closed 2018-05-03 by neu-rah