neu-rah/PCINT #4: no A0

tarbear123, 2018-05-03:
i think that you should add the arduino uno and add pins A0 to A5 (like on the arduino uno)

neu-rah, 2018-05-03:
Arduino Uno is an ATMega328, and we have the 3 ports

port B {8,9,10,11,12,13,-1,-1},
port C {14,15,16,17,18,19,20,21},
port D {0,1,2,3,4,5,6,7}

the A0 must be pin 14 and so on...
port C pins 20 and 21 are only available on arduino nano (8 analog pins)

tarbear123, 2018-05-03:
do i need to add a interrupt to the code if i have pcint.h?

neu-rah, 2018-05-03:
you need a function to be called by the interrupt
and attach it to the pin on setup
`PCattachInterrupt<pin>(myFunction,CHANGE);`

just, after that your function will be called whenever the pin CHANGEs state

tarbear123, 2018-05-03:
when i go to upload the code, i get
Arduino: 1.8.5 (Windows Store 1.8.10.0) (Windows 10), Board: "Arduino/Genuino Uno"

In file included from C:\Users\taran\Documents\Arduino\sketch_may02a\sketch_may02a.ino:1:0:

C:\Users\taran\Documents\Arduino\libraries\PCINT-master\src/pcint.h:50:5: warning: #warning using maps! [-Wcpp]

    #warning using maps!

     ^

C:\Users\taran\Documents\Arduino\libraries\PCINT-master\src/pcint.h:53:113: warning: narrowing conversion of '-1' from 'int' to 'const uint8_t {aka const unsigned char}' inside { } [-Wnarrowing]

     const uint8_t pcintPinMap[3][8] PROGMEM={{8,9,10,11,12,13,-1,-1},{14,15,16,17,18,19,20,21},{0,1,2,3,4,5,6,7}};

                                                                                                                 ^

C:\Users\taran\Documents\Arduino\libraries\PCINT-master\src/pcint.h:53:113: warning: narrowing conversion of '-1' from 'int' to 'const uint8_t {aka const unsigned char}' inside { } [-Wnarrowing]

C:\Users\taran\Documents\Arduino\sketch_may02a\sketch_may02a.ino: In function 'void setup()':

sketch_may02a:15: error: no matching function for call to 'PCattachInterrupt(int&, int)'

 PCattachInterrupt<14>(mydiceFunction,CHANGE);

                                            ^

C:\Users\taran\Documents\Arduino\sketch_may02a\sketch_may02a.ino:15:44: note: candidate is:

In file included from C:\Users\taran\Documents\Arduino\sketch_may02a\sketch_may02a.ino:1:0:

C:\Users\taran\Documents\Arduino\libraries\PCINT-master\src/pcint.h:103:7: note: template<unsigned char PIN> void PCattachInterrupt(mixHandler, uint8_t)

  void PCattachInterrupt(HANDLER_TYPE userFunc, uint8_t mode) {

       ^

C:\Users\taran\Documents\Arduino\libraries\PCINT-master\src/pcint.h:103:7: note:   template argument deduction/substitution failed:

C:\Users\taran\Documents\Arduino\sketch_may02a\sketch_may02a.ino:15:44: note:   cannot convert 'mydiceFunction' (type 'int') to type 'mixHandler'

 PCattachInterrupt<14>(mydiceFunction,CHANGE);

                                            ^

exit status 1
no matching function for call to 'PCattachInterrupt(int&, int)'

and do i need
ISR (PCINT1_vect) // handle pin change interrupt for A0 to A5 here
 {
     digitalWrite(14,analogRead(A0));
 }

neu-rah, 2018-05-03:
>sketch_may02a:15: error: no matching function for call to 'PCattachInterrupt(int&, int)'
PCattachInterrupt<14>(mydiceFunction,CHANGE);

it seems that you are passing an `int&` where you should be passing a function

can you run the example?

tarbear123, 2018-05-03:
yes i can

neu-rah, 2018-05-03:
so `mydiceFunction` should be a function like:

```c++
void mydiceFunction() {
  //... do somestuff here
}
```

tarbear123, 2018-05-03:
do i put the if(digitalRead(buttonPin) == 0 && lastButtonState == 1); //if button is pressed and was released last change else if(digitalRead(buttonPin) == 1 && lastButtonState == 0) //if button is not pressed, and was pressed last change { lastButtonState = 1; //record the lastButtonState
into the void mydicefunction

tarbear123, 2018-05-04:
when i go to upload the code to the arduino, it comes up with a compiling error and it doesn.t do that with any other code, plz help