neu-rah/PCINT #5: ESP8266 watchdog timer reset in PCattachInterrupt

jarkman, 2018-06-29:
I'm using ArduinoMenu with PCINT and an encoder on a Wemos D1 Mini Pro with an OLED shield

I'm getting a watchdog timer reset out of the 
encoder.begin();
line in setup, and with a bit of rudimentary tracing have tracked it down to 
PCattachInterrupt<pinA>(mixHandler((void(*)(void*))encoderInUpdateA,this), CHANGE);
in encoderIn.begin()

I can't see any way for that to hang, so I am quite mystified. Any clues would be very welcome.

Project is attached, it's no more than the U8G2 ArduinoMenu example with the encoder bits turned on. 

Thanks!
Richard

[ServoTester.zip](https://github.com/neu-rah/PCINT/files/2150582/ServoTester.zip)

neu-rah, 2018-06-29:
Please use alternative clickEncoder with esp8266. PCINT is for AVR's and has not been tested for ESP's

jarkman, 2018-06-29:
Thanks, will take a look.

jarkman, 2018-06-29:
Would it be worth changing the library.properties architectures line, just for clarity ?

closed 2018-06-29 by jarkman

jarkman, 2018-06-30:
Just fyi, ClickEncoder doesn't build for ESP (uses AVR #includes), and its example relies on Timer1 which is very AVR specific. Patching out the bad #includes leaves me with a sketch that always fails with a watchdog timer on start.

neu-rah, 2018-06-30:
this issue was (and still is, after some annoying scroll) on gitter room, pasting here some transcription:

**from gitter room**

it will require this include
```c++
  extern "C" {//esp8266 stuff
    #include "user_interface.h"
  }
```
creating global objects for clicEncoder, its menu driver and the esp timer and also the timer callbasck funtion
```
ClickEncoder clickEncoder(encA,encB,encBtn);
ClickEncoderStream encStream(clickEncoder,1);
os_timer_t myTimer;
void timerCallback(void *) {clickEncoder.service();}
```

on setup function, wire the timer to encoder service
```c++
    os_timer_setfn(&myTimer, timerCallback, NULL);//set the callback funtion
    os_timer_arm(&myTimer, 1, true);//setup the timer tick
```

Please let me know if i'm missing something while pasting.
Sorry for this info not yet on examples and documented, but pull-requests are very welcome also.

jarkman, 2018-06-30:
Thanks for your help. 

I've tried that, but ended up with a sketch that failed to start, apparently before setup(), with a wdt every time.

My sketch is attached if you're interested. Turning off the encoder support by removing #define USE_ENCODER results in a sketch that runs & displays, so the problem must be encoder-related.

I can't see any obvious cause of that problem. 

[ServoTesterBadMenus.zip](https://github.com/neu-rah/PCINT/files/2151648/ServoTesterBadMenus.zip)

neu-rah, 2018-06-30:
you have
```c++
#define encA 5
```
pin 5 is a I2C port (at least on my esp board), please ensure that you are not using that port for your lcd

neu-rah, 2018-06-30:
here using pins 12,13,14 for encoder (wemos D1 mini)

```c++
#include <Arduino.h>

/********************
Arduino generic menu system
U8G2 menu example
U8G2: https://github.com/olikraus/u8g2

Oct. 2016 Stephen Denne https://github.com/datacute
Based on example from Rui Azevedo - ruihfazevedo(@rrob@)gmail.com
Original from: https://github.com/christophepersoz

menu on U8G2 device
output: Wemos D1 mini OLED Shield (SSD1306 64x48 I2C) + Serial
input: Serial + encoder
mcu: nano328p

*/

extern "C" {
#include "user_interface.h"
}

#include <menu.h>
#include <menuIO/u8g2Out.h>

  extern "C" {//esp8266 stuff
    #include "user_interface.h"
  }

#define USE_ENCODER

#ifdef USE_ENCODER
#include <ClickEncoder.h>
#include <menuIO/clickEncoderIn.h>
#endif  

#include <menuIO/keyIn.h>
#include <menuIO/chainStream.h>
#include <menuIO/serialOut.h>
#include <menuIO/serialIn.h>

using namespace Menu;

#define LEDPIN LED_BUILTIN

//#define encA 5
//#define encB 6
//this encoder has a button here
//#define encBtn 7 //D0

// #define USE_PCD8544
#define USE_SSD1306

#if defined(USE_PCD8544)
  // rotary encoder pins
  // #define encA    2
  // #define encB    3
  // #define encBtn  4

  #include <SPI.h>
  #define USE_HWSPI
  #define U8_DC 9
  #define U8_CS 8
  #define U8_RST 7
  #define fontName u8g2_font_5x7_tf
  #define fontX 5
  #define fontY 9
[... 199 lines cut ...]
  mainMenu[1].enabled=disabledStatus;
  nav.idleTask=idle;//point a function to be used when menu is suspended

#ifdef USE_ENCODER
  os_timer_setfn(&myTimer, timerCallback, NULL);//set the callback funtion
  os_timer_arm(&myTimer, 1, true);//setup the timer tick
#endif
    
  Serial.println("setup done.");Serial.flush();
}

void loop() {
  nav.doInput();
  // digitalWrite(LEDPIN, ledCtrl);
  if (nav.changed(0)) {//only draw if menu changed for gfx device
    //change checking leaves more time for other tasks
    u8g2.firstPage();
    do nav.doOutput(); while(u8g2.nextPage());
  }
  delay(100);//simulate other tasks delay
}
```

this code seems stable, menu shown on screen, no reset.
was not able to test with real encoder attached (no encoder at hand right now) neither with serial as your code does not allow simultaneous input from Serial and encoder

jarkman, 2018-07-02:
Mine is a 1.1:
https://wiki.wemos.cc/products:retired:d1_mini_pro_v1.1.0
with I2C on D1 & D2.

But, thank you for pointing the finger of suspicion at pin numbering. What I had missed (and please excuse the prolonged screaming noise here) is that (eg) D5 is not 5. Once I went to this:
#define encA D5
#define encB D6
#define encBtn D7
it all burst into life.

Once again, thanks for your help and your lovely library. I'm off to actually make my thing now! :-)