neu-rah/PCINT #11: pcint.h: error: 'INT_MODE_TYPE' has not been declared

KKuKK, 2024-06-25:
Hi 
I tried to use your library with an MEGA2560. I tried to compile the example btnled.ino and it stops compiling with following error:

"PCINT\src/pcint.h:108:48: error: 'INT_MODE_TYPE' has not been declared
  void PCattachInterrupt(HANDLER_TYPE userFunc, INT_MODE_TYPE mode) {"

I am using Arduino IDE 1.8.57.0

I found this issue in an articlle which says

"It looks like when adding support for STM32F1 and STM32F4 processors they had to make the RISING/FALLING/CHANGE argument something other than 'uint8_t'. They did that by defining "INT_MODE_TYPE" but it seems they only do that when "PCINT_NO_MAPS" is defined. I don't think they tested on a MEGA after making this change."

Is there any solution? I think this lib is easy to use. There are two ohters which I cannot handle.
Thanks for your help.
Greatings Klaus

rossisbudda, 2025-01-22:
Hi! I have the same issue with an Arduino Nano Every and IDE2.3.4. It's affecting the encoder stuff on arduinoMenu which is how I encountered it.

StormingMoose, 2025-05-24:
Same issue on Arduino Uno.